DCC-EX/EX-IOExpander #12: Bug Report: displaying wrong version

momfertje, 2022-12-21:
### Version

v0.0.3 alpha

### Bug description

When issuing <D HAL SHOW> serial returns wrong version:

`<* EX-IOExpander I2C:x65 Configured on Vpins:800-817  *>
**<* EX-IOExpander x65 using driver version IN *>**
<* EX-IOExpander x65: Digital Vpins 800-811, Analogue Vpins 812-817 *>
<* Arduino Vpins:2-69 *>` 

### Steps to reproduce the bug

issue <D SHOW HAL> in Serial Monitor (CS Side)

### Expected behaviour

should display the correct EX_IOexpander version

### Hardware in use

Mega Clone with DEEK ROBOT Motor Shield Special CS version 4.2.7pre 

Nano (old bootloader) 0.0.3 EX_IOepxander

peteGSX, 2022-12-21:
Not sure why this displays "IN" rather than the version, however have disabled the device driver version display for the moment.

Contemplating reading the EX-IOExpander software version from the device to display instead, will leave this open for the moment.

peteGSX, 2022-12-25:
Closed by https://github.com/DCC-EX/EX-IOExpander/pull/16

closed 2022-12-25 by peteGSX