DCC-EX/EX-IOExpander #9: Compilation fails for Mega

Rosscoetrain, 2022-12-21:
### Version

pre1EXIoexpander

### Bug description

Compiling fails with:

Arduino: 1.8.19 (Linux), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

EX-IOExpander:48:1: error: expected ',' or ';' before 'typedef'
 typedef struct {
 ^~~~~~~
EX-IOExpander:52:3: error: 'digitalConfig' does not name a type; did you mean 'digitalPinMap'?
 } digitalConfig;
   ^~~~~~~~~~~~~
   digitalPinMap
EX-IOExpander:67:1: error: 'digitalConfig' does not name a type; did you mean 'digitalPinMap'?
 digitalConfig digitalPins[NUMBER_OF_DIGITAL_PINS + NUMBER_OF_ANALOGUE_PINS];
 ^~~~~~~~~~~~~
 digitalPinMap
/home/ross/Pictures/marklin/electronics/arduino/EX-IOExpander/EX-IOExpander.ino: In function 'void loop()':
EX-IOExpander:131:11: error: 'digitalPins' was not declared in this scope
       if (digitalPins[dPin].direction == 1) {
           ^~~~~~~~~~~
/home/ross/Pictures/marklin/electronics/arduino/EX-IOExpander/EX-IOExpander.ino:131:11: note: suggested alternative: 'digitalPinMap'
       if (digitalPins[dPin].direction == 1) {
           ^~~~~~~~~~~
           digitalPinMap
/home/ross/Pictures/marklin/electronics/arduino/EX-IOExpander/EX-IOExpander.ino: In function 'void receiveEvent(int)':
EX-IOExpander:177:13: error: 'digitalPins' was not declared in this scope
             digitalPins[pin].direction = 0;
             ^~~~~~~~~~~
/home/ross/Pictures/marklin/electronics/arduino/EX-IOExpander/EX-IOExpander.ino:177:13: note: suggested alternative: 'digitalPinMap'
             digitalPins[pin].direction = 0;
             ^~~~~~~~~~~
             digitalPinMap
EX-IOExpander:211:9: error: 'digitalPins' was not declared in this scope
         digitalPins[pin].direction = 1;   // Must be an input if we got a pullup config
         ^~~~~~~~~~~
/home/ross/Pictures/marklin/electronics/arduino/EX-IOExpander/EX-IOExpander.ino:211:9: note: suggested alternative: 'digitalPinMap'
         digitalPins[pin].direction = 1;   // Must be an input if we got a pullup config
         ^~~~~~~~~~~
         digitalPinMap
EX-IOExpander:244:9: error: 'digitalPins' was not declared in this scope
         digitalPins[dPin].direction = 0;
         ^~~~~~~~~~~
/home/ross/Pictures/marklin/electronics/arduino/EX-IOExpander/EX-IOExpander.ino:244:9: note: suggested alternative: 'digitalPinMap'
         digitalPins[dPin].direction = 0;
         ^~~~~~~~~~~
         digitalPinMap
EX-IOExpander:253:31: error: 'digitalPins' was not declared in this scope
         digitalOutBuffer[0] = digitalPins[dPin].state;
                               ^~~~~~~~~~~
/home/ross/Pictures/marklin/electronics/arduino/EX-IOExpander/EX-IOExpander.ino:253:31: note: suggested alternative: 'digitalPinMap'
         digitalOutBuffer[0] = digitalPins[dPin].state;
                               ^~~~~~~~~~~
                               digitalPinMap
exit status 1
expected ',' or ';' before 'typedef'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

**Needs a ; at line 59 in SupportedDevices.h**

### Steps to reproduce the bug

1. Compile in arduino ide with board selected as Arduino Mega 2560
2. 

### Expected behaviour

Should compile correctly with ; at line 59 in SupportedDevices.h

### Hardware in use

EX-IOExpander arduino mega 2560

momfertje, 2022-12-21:
To solve the issue: add a ; at the end of line 50 of  SupportedDevices.h

peteGSX, 2022-12-21:
Ah yeah I missed that! I haven't tried to compile for Mega yet so I didn't see that missing comma.

I'll address this along with the bug fix for detecting the wrong board.

closed 2022-12-21 by peteGSX