DCC-EX/EX-IOExpander #51: Bug Report: Variable not declard

martinfantini, 2023-03-19:
### Version

HEAD

### Bug description

The variable is not declared as [bool](https://github.com/DCC-EX/EX-IOExpander/blob/main/EX-IOExpander.ino#L81) :

```
#ifdef DIAG
  diag = true;
#else
  bool diag = false;
#endif
```

The solution is:

```
#ifdef DIAG
  bool diag = true;
#else
  bool diag = false;
#endif
```

### Steps to reproduce the bug

Not compile if you enable the `DIAG` compilation variable.

### Expected behaviour

Compile. 

### Hardware in use

EX-IOExpander Arduino Mega

peteGSX, 2023-03-19:
Will fix this along with https://github.com/DCC-EX/EX-IOExpander/issues/49

martinfantini, 2023-03-19:
Nice. Thanks! I will glad to do the code review.

closed 2023-03-20 by peteGSX