DCC-EX/EX-IOExpander #52: Bug Report: Note declared

martinfantini, 2023-03-19:
### Version

HEAD

### Bug description

Not declared variables of I2C fro Arduino mega.

Solution: In the file **arduino_avr_mega.h**, you have to add this  definition variables:

```
#define I2C_SDA 20
#define I2C_SCL 21
```

### Steps to reproduce the bug

1. Run in Arduino Mega the Ex pander.
2. Declare som VPins
3. Always apear the error in the central: 

`Vpin XXX cannot be used as a digital input pin`

### Expected behaviour

Declared the VIO pins.

### Hardware in use

Arduino Mega

peteGSX, 2023-03-19:
I don't have a Mega available to test this on at the moment, but I'm curious to see how not defining these relates to the error `Vpin XXX cannot be used as a digital input pin`.

The only reason these need to be defined is if the I2C pullups are internal only and can be disabled. In the case of the Mega these are external and cannot be disabled, therefore these aren't defined for the Mega platform.

I'll have to see if I can replicate the behaviour by commenting these out on an Uno or Nano.

Also, do you have `#define DISABLE_I2C_PULLUPS` commented out in myConfig.h?

martinfantini, 2023-03-19:
I have an sync error, and this error appears on the [DCC-Command station](https://dcc-ex.com/ex-commandstation/index.html). 

_Also, do you have #define DISABLE_I2C_PULLUPS commented out in myConfig.h?_
Yes. I did that and after that appeared the map with the pins.

I have connected to the I2C of DCC-Command station also the display and the EX-IOExpander with mega extension. But for some reason, when I start the Mega, there is a sync error. 
 - Do you know if the I2C could not work because there is more then one component connected?
 - Is it possible to have more than one extension connected to the central?

peteGSX, 2023-03-19:
You may need to try using some external pull up resistors perhaps? I believe 4k7 is a good value to start with.

You can have multiple devices on the I2C bus, and my testing is normally done with an OLED plus MCP23017, PCA9685, as well as an Uno and Nano running EX-IOExpander.

Are you able to share the exact error message for the sync error?

martinfantini, 2023-03-19:
Yes. I have this configutation on the EC-IOExtender:

```
Received correct pin count: 62, starting at Vpin: 800
Vpin to physical pin mappings (Vpin => physical pin):
|800 =>   2|801 =>   3|802 =>   4|803 =>   5|804 =>   6|805 =>   7|806 =>   8|807 =>   9|
|808 =>  10|809 =>  11|810 =>  12|811 =>  13|812 =>  14|813 =>  15|814 =>  16|815 =>  17|
|816 =>  18|817 =>  19|818 =>  22|819 =>  23|820 =>  24|821 =>  25|822 =>  26|823 =>  27|
|824 =>  28|825 =>  28|826 =>  30|827 =>  31|828 =>  32|829 =>  33|830 =>  34|831 =>  35|
|832 =>  36|833 =>  37|834 =>  38|835 =>  39|836 =>  40|837 =>  41|838 =>  42|839 =>  43|
|840 =>  44|841 =>  45|842 =>  46|843 =>  47|844 =>  48|845 =>  49|846 =>  54|847 =>  55|
|848 =>  56|849 =>  57|850 =>  58|851 =>  59|852 =>  60|853 =>  61|854 =>  62|855 =>  63|
|856 =>  64|857 =>  65|858 =>  66|859 =>  67|860 =>  68|861 =>  69|
Digital Pin|Direction|Pullup|State:2|1|1|0
Digital Pin|Direction|Pullup|State:3|1|1|0
Digital Pin|Direction|Pullup|State:4|1|1|0
```

Which means the Pin configuration is a Pull-Up (which in those Pins , I have the 5V) but I put them with  0V. This exmaple with the Pin 2, I do not have the feedback in the Mega:

```
Pin 68 not in use
Pin 69 not in use
Digital Pin|Direction|Pullup|State:2|1|1|1
Digital Pin|Direction|Pullup|State:3|1|1|0
Digital Pin|Direction|Pullup|State:4|1|1|0
Pin 5 not in use
Pin 6 not in use
Pin 7 not in use
```

This NOT happen, if i configure the PIN 22 of the DCC-Central

martinfantini, 2023-03-19:
Adding the JRMI logs:

```
23:21:33.389: [0]  TX: Track Power OFF Cmd 
23:21:33.397: [p0]  RX: Power Status: OFF
23:21:36.998: [Q 22]  RX: Sensor Reply (Active): Number: 22, State: ACTIVE
23:21:37.017: [q 22]  RX: Sensor Reply (Inactive): Number: 22, State: INACTIVE
23:21:37.048: [Q 22]  RX: Sensor Reply (Active): Number: 22, State: ACTIVE
23:21:38.746: [q 22]  RX: Sensor Reply (Inactive): Number: 22, State: INACTIVE
23:21:48.835: [s]  TX: Status Cmd 
23:21:48.883: [iDCC-EX V-4.2.29 / MEGA / STANDARD_MOTOR_SHIELD G-devel-202303151539Z]  RX: Status:Station: DCC-EX, Build: devel-202303151539Z, Version: 4.2.29
23:21:48.887: [p0]  RX: Power Status: OFF
23:21:48.893: [q 800]  RX: Sensor Reply (Inactive): Number: 800, State: INACTIVE
23:21:48.897: [q 24]  RX: Sensor Reply (Inactive): Number: 24, State: INACTIVE
23:21:48.901: [q 22]  RX: Sensor Reply (Inactive): Number: 22, State: INACTIVE
23:21:48.905: [q 801]  RX: Sensor Reply (Inactive): Number: 801, State: INACTIVE
23:21:48.910: [q 818]  RX: Sensor Reply (Inactive): Number: 818, State: INACTIVE
23:21:48.915: [q 802]  RX: Sensor Reply (Inactive): Number: 802, State: INACTIVE
```

I've never seen the ACTIVE status in those sensors (800/801/802/818).

martinfantini, 2023-03-26:
I put the two Pull-up resistors and update the DCC Command station and the DCC-Extender. and it starts working.

closed 2023-03-26 by martinfantini

peteGSX, 2023-03-26:
Glad you got it working! I have ordered a Mega and will have it soon so I can test things for myself as well.

martinfantini, 2023-03-27:
My connection was adding the two 4k7 pull-up resistors, but I have to try without them. Because the board has the resistors inside.