DCC-EX/EX-IOExpander #55: Bug Report: The Outputs A10 to A15 not working

martinfantini, 2023-04-14:
### Version

HEAD

### Bug description

The outputs A10 to A15 in Arduino Mega are not Working. The log in the JMRI is:

```
10:59:06.656: [Q 861]  RX: Sensor Reply (Active): Number: 861, State: ACTIVE
10:59:06.660: [Q 860]  RX: Sensor Reply (Active): Number: 860, State: ACTIVE
10:59:06.664: [Q 859]  RX: Sensor Reply (Active): Number: 859, State: ACTIVE
10:59:06.669: [Q 858]  RX: Sensor Reply (Active): Number: 858, State: ACTIVE
10:59:06.673: [Q 857]  RX: Sensor Reply (Active): Number: 857, State: ACTIVE
10:59:06.677: [Q 856]  RX: Sensor Reply (Active): Number: 856, State: ACTIVE
```

and the outputs are not active.

### Steps to reproduce the bug

1. Set up the outputs from the [table](https://dcc-ex.com/ex-ioexpander/supported-devices.html) with pull up resistors for Arduino Mega.
2. Leave the outputs without any connection. (They will appear as ACTIVE)
4. Put the output with GND connection, to simulate the input.

### Expected behaviour

In the case of the report, those outputs have to be INACTIVE. 

```
10:59:06.656: [Q 861]  RX: Sensor Reply (Active): Number: 861, State: ACTIVE
10:59:06.660: [Q 860]  RX: Sensor Reply (Active): Number: 860, State: ACTIVE
10:59:06.664: [Q 859]  RX: Sensor Reply (Active): Number: 859, State: ACTIVE
10:59:06.669: [Q 858]  RX: Sensor Reply (Active): Number: 858, State: ACTIVE
10:59:06.673: [Q 857]  RX: Sensor Reply (Active): Number: 857, State: ACTIVE
10:59:06.677: [Q 856]  RX: Sensor Reply (Active): Number: 856, State: ACTIVE
```

As soon as I connect to them to GND, they has to change to ACTIVE.

### Hardware in use

EX-IOExpander (Arduino Mega) + Central (Arduino Mega + Motor Shield + Arduino Netwok Shield )

peteGSX, 2023-04-15:
Thanks for reporting @martinfantini and even more so for finding the fix! Yes, that is an obvious oversight on my part and will not work for the higher numbered pins as inputs.

I'll fix this in an upcoming update, not sure when just yet though.

martinfantini, 2023-04-16:
You welcome! Please @peteGSX put this line: 

`digitalPinBytes = (numDigitalPins + 7) / 8;`

I got this from the central, from [here](https://github.com/DCC-EX/CommandStation-EX/blob/devel/IO_EXIOExpander.h#L101).

peteGSX, 2023-04-17:
@martinfantini give 0.0.20 a try.

closed 2023-04-17 by peteGSX

martinfantini, 2023-04-17:
@peteGSX Thanks! I will do it.